bluerobotics/navigator-rs #5: Error at first use of each pin from GPIO

RaulTrombin, 2023-05-19:
We are using 5 GPIOs as output, when fresh boot a rapsberry, you have one error for each, then it acts normal.
Check bellow when ./example is called:

```
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
thread '<unnamed>' panicked at 'Error: Setting CS2 pin as output: Io(Os { code: 13, kind: PermissionDenied, message: "Permission denied" })', /cargo/git/checkouts/navigator-rs-2872844b08e6ceff/eefa796/src/[lib.rs:129](http://lib.rs:129/):14
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 9
Aborted
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
thread '<unnamed>' panicked at 'Error: Setting CS2 pin as output: Io(Os { code: 13, kind: PermissionDenied, message: "Permission denied" })', /cargo/git/checkouts/navigator-rs-2872844b08e6ceff/eefa796/src/[lib.rs:135](http://lib.rs:135/):14
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 9
Aborted
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
thread '<unnamed>' panicked at 'Error: Setting led pins as output: Io(Os { code: 13, kind: PermissionDenied, message: "Permission denied" })', /cargo/git/checkouts/navigator-rs-2872844b08e6ceff/eefa796/src/[lib.rs:68](http://lib.rs:68/):18
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 9
Aborted
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
thread '<unnamed>' panicked at 'Error: Setting led pins as output: Io(Os { code: 13, kind: PermissionDenied, message: "Permission denied" })', /cargo/git/checkouts/navigator-rs-2872844b08e6ceff/eefa796/src/[lib.rs:68](http://lib.rs:68/):18
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 9
Aborted
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
thread '<unnamed>' panicked at 'Error: Setting led pins as output: Io(Os { code: 13, kind: PermissionDenied, message: "Permission denied" })', /cargo/git/checkouts/navigator-rs-2872844b08e6ceff/eefa796/src/[lib.rs:68](http://lib.rs:68/):18
note: run with RUST_BACKTRACE=1 environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 9
Aborted
pi@blueos:~/navigator-lib/examples/cpp/build $ ./example 
Navigator C test, system details:
System is Linux on armv7l hardware
OS Release is 5.10.33-v7l+
OS Version is #1415 SMP Fri Apr 30 15:50:57 BST 2021
Iniciating navigator module.
Setting leds on!
Temperature: 39.639999
Pressure: 101.837517
Reading ADC Channels: 1 = 410, 2 = 418, 3 = 427, 4 = 316
Magnetic field: X = 3.600000, Y = -12.150001, Z = 18.750000
Acceleration: X = 0.002394, Y = -0.010176, Z = -0.308264
Gyroscope: X = -0.001198, Y = 0.002131, Z = 0.000399
Setting leds off!
```

patrickelectric, 2023-05-19:
do you have BlueOS running ? If so, is it running ArduPilot ? If so, set it to run SITL on the Ardupilot Manager.

RaulTrombin, 2023-06-14:
Reported on [linux-embedded-hal](https://github.com/rust-embedded/linux-embedded-hal/issues/93)

joaoantoniocardoso, 2023-06-14:
Solved by #9

closed 2023-06-14 by joaoantoniocardoso