bluerobotics/navigator-rs #42: Add support to detected sensors

patrickelectric, 2023-08-07:
Add automatic detection for Bar30 and Bar100

RaulTrombin, 2023-09-26:
@patrickelectric @ES-Alexander 
Will we add then to navigator-rs or can create a separate crate project?
Can we add these components to parts list?

- [ ] take parts

ES-Alexander, 2023-10-11:
How well does rust deal with dynamic libraries?

I feel like ideally they would be optional components since the sensors don’t come built in, and we already have libraries for them elsewhere that are maintained independently. If it’s not hard for a user to choose with/without the extra sensor libraries then that would be best, but if that would be challenging then we might need to do a monolithic crate with everything in it 🤷‍♂️

patrickelectric, 2023-10-11:
Adding support for such sensors is not a big deal, we can just add in navigator-rs.

RaulTrombin, 2024-04-25:
I don't see how it'll work. 
Will it detect a device address on the I2C bus? 
The devices need their own crates/examples beforehand.