bluerobotics/navigator-rs #54: Issue: ADC Measuring time

RaulTrombin, 2023-10-18:
actually it takes 40.014108 ms to read the 4 ADC channels,
need to check if we can speedup this.

          {
            "name": "read_adc_all",
            "value": 40014108,
            "range": "± 292315",
            "unit": "ns/iter"
          },
           
[image]

patrickelectric, 2023-10-18:
From the datasheet:
> The ADS111x perform conversions at data rates up
to 860 samples per second (SPS).

ref: https://www.ti.com/lit/ds/symlink/ads1115.pdf

RaulTrombin, 2023-10-20:
found the method to change it. 
set_data_rate(DataRate16Bit::Sps860);

```
read_adc                time:   [3.0672 ms 3.0759 ms 3.0840 ms]
                        change: [-69.244% -69.143% -69.034%] (p = 0.00 < 0.05)
                        Performance has improved.
Found 23 outliers among 100 measurements (23.00%)
  17 (17.00%) low severe
  1 (1.00%) low mild
  4 (4.00%) high mild
  1 (1.00%) high severe

read_adc_all            time:   [12.350 ms 12.375 ms 12.409 ms]
                        change: [-69.174% -69.086% -68.976%] (p = 0.00 < 0.05)
                        Performance has improved.
Found 23 outliers among 100 measurements (23.00%)
  14 (14.00%) low severe
  1 (1.00%) low mild
  3 (3.00%) high mild
  5 (5.00%) high severe
```

Already worth to open a PR? Or we can check it further

closed 2023-11-01 by RaulTrombin