bluerobotics/navigator-rs #77: Implementation Hyperlink is broken?

hikari20XX, 2024-04-29:
Hello,

I'm having trouble accessing the Implementations documentation through the hyperlink, is there another place I can get to it? Whenever I try to run the example scripts or scripts that require the navigator_rs crate, I get the error: 
unresolved import `navigator_rs`
 --> rainbow.rs:1:5
  |
1 | use navigator_rs::Navigator;
  |     ^^^^^^^^^^^^ maybe a missing crate `navigator_rs`?
  |
  = help: consider adding `extern crate navigator_rs` to use the `navigator_rs` crate

error: aborting due to 1 previous error

Any advice/wisdom is greatly appreciated.

RaulTrombin, 2024-05-16:
Hey @hikari20XX ,
Are you running from this repository ?
`cargo run --example rainbow`

Or are you importing this library to your custom project?
Then you need:
`cargo install navigator-rs`

https://docs.bluerobotics.com/navigator-rs/doc/navigator_rs/struct.Navigator.html

RaulTrombin, 2024-05-16:
@hikari20XX 
https://github.com/bluerobotics/navigator-rs/assets/80598030/9e413fa7-1a6b-4823-902d-dec04439fa23