bluerobotics/navigator-rs #88: Delay after pin export seems not long enough

fnauert, 2024-11-27:
Hi there :)
I'm running the navigator  lib on a raspberry pi4 with Ubuntu 20.04. When I execute the simple example I sometimes face permission denied errors, since the library can not access the gpio pins. However, the pin is exported and if I start the executable again it works fine. I tried setting the delay after the pin export to 60 milliseconds instead of 30  in my local navigator-rs library and that fixed the issue for me. Would it be possible to increase the delay in general, so that I can use the official library release?

patrickelectric, 2024-11-27:
Hi @fnauert thanks for raising this issue, It's the first time that I'm seeing this error. Can you confirm after a reboot that moving to 60 it indeed fixes the problem ?

fnauert, 2024-11-28:
Hi @patrickelectric yes, I rebooted at least three times since I increased the delay and the error did not occur again.

fnauert, 2024-11-28:
Thank you very much for the fast support :heart:

PR #91 (2024-11-28): {icm20689, leak, led, pca9685}: Move from 30 to 60ms to wait pin configuration
Fix #88